xC0000005/Marlin #1: Build Help Thread

xC0000005, 2019-05-29:
Use this to ask questions about building/configuration.

atomgomba, 2019-06-17:
I just compiled and flashed (previously was stock V41) from e03432b and I did not get the message on the LCD with version number "V99". Even though I see this response in OctoPrint:
```
Recv: FIRMWARE_NAME:Marlin bugfix-2.0.x (Github) SOURCE_CODE_URL:https://github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:Malyan M200 EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff
```
And this seems to be success regardless.

xC0000005, 2019-06-17:
In configuration.h, is MALYNA_LCD enabled?

atomgomba, 2019-06-17:
I have a clean working tree, same state as remote.

xC0000005, 2019-06-17:
After you flash the first time, try removing the SD card and restarting. The UI only waits a limited amount of time and flashing often takes longer.

xC0000005, 2019-06-20:
Another possibility is that you have the X and Y limit switches closed, so the bootloader is trying to update the UI firmware. Remove the SD card, move X and Y so they're not closed, and restart, see if it works.

tthfry, 2019-08-18:
I've started messing with marlin again and used your portable package for building. Worked like a charm, but every second time you launch the IDE, the STM32 boards are gone. If you download them from the repository, select "3D Printer Boards" and "Malyan M200" (in my case v1), upon building it says that it can't open printer boards because they're not found.

It appears that opening the board manager alone is enough to let it "forget" the old STM library.

xC0000005, 2019-08-19:
I’ll look into this - ti’s probably some bug in Arduino, and I’ll run it down if I can. As of a day or so ago, Marlin Main should generally speaking “just work” for the M200. The readme.md in the Malyan config directory is really the prereq - ever since 1.6.0 of the ST Arduino core, M200 support has just worked for V1 and V2. There’s some work that needs to happen for V3 and late V2 failsave (and M300) but V1/V1 early run/V2 mid run should “just work”.

tthfry, 2019-08-24:
Okay, tried some more. I can build with STM 1.6.1 and latest marlin bugfix branch when using the 1.8.9 IDE. I can't build using the beta 1.9 IDE from the portable package and the included STM (I guess it's 1.6.0-dev?) once I opened the board manager or closed and reopened the IDE. In the latter case I get the "board not found" error. So it seems to be just the portable version.

xC0000005, 2019-08-24:
Ok. I can repro this but I haven’t had a chance to debug the IDE to see what the heck it’s doing.

Eno423, 2019-10-21:
anyupdate on this? I tried a few diff version of the SMT32 core, but all give the same board 3dprinter is unknown in IDE 1.90

fritztoch, 2020-01-04:
I am stuck at Step 4. I have downloaded Marlin 2.x, and I have downloaded M200Fixes. Am I supposed to merge the two folders? The instructions don't say what to do with M200Fixes.

theodonkulus, 2020-01-07:
Hi has this been fully tested on Marlin 2.0 or does this only work on the bugfix 2.0 branch? Ive got a M200 V2 and when I upgrade the firmware with the example config the system keeps resetting and I'm unable to control the motors from the LCD screen.

xC0000005, 2020-01-08:
It fails on 2.0 due to a stepper issue I haven't had time to debug (but oddly, the M300 appears to work??).

fritztoch, 2020-01-11:
Got a successful build with STM32 1.6.1 and Arduino 1.8.10 (and remembering to set the C Runtime Library to Printf). Web interface won't move any axis but can preheat. Front panel controls work for preheat, move E and Z. Home X & Y works too but is slow. This is a V2.

theodonkulus, 2020-02-01:
So I should just build the M300 binary then for the V2? Would that be okay?

xC0000005, 2020-02-01:
@theodonkulus, build for whichever board you have. If you're using the most recent (github) STM32 core, you'll note there's a board definition for the M300 now (thanks to @mojocorp). If you're building for the M300 (delta) you should use that (and probably his pins, since I'm sure some of mine were wrong). If you're building for the V2, select V2, V1, V1 and so on. The only oddity are the very early M200 V2s - the first thousand or so have STM32F103 MCUs, BUT have different steps per MM and the V2 display/fan/endstop connectors. For those, you select V1 and set your steps per mm right. The early V2s are also distinctly louder than most V1/V2s.

theodonkulus, 2020-02-01:
How would I know which one I have in terms of the new or old V2s I didn't think there was a Gcode command that would let me view the chips P/N? I couldnt seem to get anything to work and had the LCD constantly resetting on me when I used the above steps for the V2 system

xC0000005, 2020-02-01:
Here’s how I look at it when figuring out what’s in a machine:
1. If there’s no fan in the rear of the base, it’s a V1, period, STM32F103
2. If there is a fan in the rear of the base and it’s white, it’s a V2, no doubt, STM32F070.
3. This is the tricky one. If there’s a fan in the bakc and it’s black, it *might* be a STM32F103, OR a 070, but it’s far more likely (by thousands) that it’s an 070.  Build for 070, and if that doesn’t work, build for 103. 
4. If it’s a delta, it’s am M300/070. The only exception is so rare I am not sure how to quantify it. There were perhaps a dozen deltas with a V2 070 mainboard - I only know this because I’ve seen a picture of it, not because I own one. These may have been pre-production prototypes, low number test runs, or something else - I can’t be sure.

You could, of course, always open the machine up and look at the MCU under a magnifying glass (or without, if you are younger and sharper-eyed than I).

mojocorp, 2020-02-03:
> @theodonkulus, build for whichever board you have. If you're using the most recent (github) STM32 core, you'll note there's a board definition for the M300 now (thanks to @mojocorp). If you're [...]

The M300 is not ready for prime time: while it's working fine with STM32duino 1.7.0, it's crashing with 1.8.0.....
Moreover, the stock M300 power supply is too weak to power the bed and nozzle at the same time. The stock firmware trick is to alternate the heaters to workaround this problem.
To resume: wait for a fix in Marlin or STM32duino for the crash, and in the mean time search for a 12V 10A power supply. (I use an xbox 360 power supply and replaced the cable)

xC0000005, 2020-02-03:
The issue is possibly that the ability to override the timers is currently broken (you’ll note the STM32F1 HAL has a #ifndef for timer numbers) Try setting timer max to 0xFFFF and timers to 1 and 3, hardcoded in timers.h.

theodonkulus, 2020-02-03:
Ill have to poke around tonight with those changes. It looks like I have a V2 as its white with the fan so its STM32F070. I was hoping it was mostly a config issue on my side but okay. Good thing I program for a living. :D. Guessing the code is stubbing out hardware timers to do PWM?

xC0000005, 2020-02-03:
It does - ST’s arduino core has no soft PWM feature, though I’ve thought several times about how we could switch to a PWM::write model. Then we could do PWM via a single timer for all pins, or simply soft PWM. I published an M200Fixes2 branch which has the timers fix, but not the EXTUI stuff we need if you print from Octoprint.

theodonkulus, 2020-02-03:
Makes sense. That would explain why steppers werent moving at all. 

Ah EXTUI is the LCD panel? 

I do plan on using octoprint with these. My goal is to add some touch sensors for the bed using 3 point method on a glass plate + filament a runout + linear advance. I saw what you did on hackaday and figured I could rig these all up once Marlin was running using the GPIO.

xC0000005, 2020-02-03:
EXTUI is the LCD, yes. Later M200 V2s and the deltas have a failsafe built in that switches the heaters off after about 30 minutes if the SYS:BUILD command isn’t issued. I have a fix for that that needed more testing in my M200Fixes branch (I don’t think I’ve pulled it over). In short, I have a sort of “On Heating” event that issues the BUILD. You can tell if you require this by issuing whatever firmware Malyan shipped you and a temp GCODE command  and watching your LCD - if it switches to the BUILD screan automatically, you need it.

theodonkulus, 2020-02-03:
Well ive got five v2s i can test this on if you have a branch to point me too. I did see my LCD behave eratically and switch screens withour button presses when I last ran the firmware along with no stepper movement. I was using these for a small scale printfarm once I modded hotend, and lead screws.

mcheah, 2020-02-03:
FYI, how malyan does it and how i do it is to send sys:build on M109 and M190 with the logic being that every print will start with that.

xC0000005, 2020-02-03:
That’s exactly what the EXTUI fix does - I added an event, and the Malyan LCD impementation in Marlin issues a SYS:BUILD. (I don’t think I ever would have figured out this was happening without your help.)

theodonkulus, 2020-02-03:
I guess its worth adding that Gcode to Cura then for my mini v2s

xC0000005, 2020-02-03:
If your minis switch to the build screen when you issue a M109 or related temperature command, you’ll need an additional fix grabbed from another branch, that’s all.

theodonkulus, 2020-02-07:
@xC0000005  What sha-1's should I be grabbing from M200Fixes? Tried a straight rebase off the marlin v2.0.1 tag where I was originally trying these changes. Or should I be building off earlier versions of marlin with your fixes?

theodonkulus, 2020-04-12:
Hi @xC0000005 Just aheads up, I finally got this to work after playing around with some settings and then reading something on reddit about the STM32 cores library. The 1.8.0 has a watchdog timer bug in it and after reverting to 1.7.0 I was able to successfully compile then upload code via SD card upgrade.  I also cherry picked your M200fixes2 branch ontop of the most recent Marlin off 2.0.5.3 and built just fine as well.

I'll be tinkering away to figure out if there's any more features I can cram into this little printer.

Thanks for your support and hope you're safe during this time.

Gunner087, 2020-05-22:
Is it possible to flash the V2 F070 board using the ST-LINK V2? I've tried flashing using the sd card slot but I cannot get it to flash, I've tried multiple sd cards and multiple file formats.

xC0000005, 2020-05-22:
It is, but you have to be comfortable soldering to some very small via pads on the board.

Gunner087, 2020-05-22:
I'm up for it, which pads is it?

Gunner087, 2020-05-22:
So I got wires soldered for SWCLK, SWDIO, and GND. Then I tried using the "STM32CubeProgrammer (SWD)" upload method in the Arduino IDE to flash the STM32 with Marlin. This successfully uploaded Marlin but also overwrote the MonoPrice factory bootloader that I didn't realize is stored at the beginning of the STM32's flash memory. Thanks to the work done by Akro [here](http://arkorobotics.com/blog/?p=161) I was able to fix that by flashing my board with his version of the MonoPrice Delta firmware, which included his custom bootloader. I then flashed the Marlin binary to the STM32 manually with the starting address set to 0x08002000 using the actual STM32CubeProgrammer application. This worked and the printer now seems to run Marlin successfully. Thought I would post this so others will know to use 0x08002000 as the starting address to avoid overwriting the factory bootloader.

xC0000005, 2020-05-22:
I’m not capable of soldering (my hands are very shaky), but I do have a sketch meant to calculate the right CRC seed and rewrite the bootloader with the right value. At some point, I’ll debug it through and publish it. The idea is that you flash it at 0x8000000, then let it calculate the right seed, patch the in-memory bootloader, and write that back to the first two pages. After that, you’re “factory original."

Gunner087, 2020-05-22:
I don't actually know that much about this kind of stuff, what does that mean in a less technical explanation? A quick google didn't help me, haha.

KillerXtreme-, 2021-06-06:
So I'm trying to compile Marlin for my MPSM(MalyanM200) early v2 that has the v1 Mainboard. When following these directions all I get is "Error compiling for board 3D printer boards." Along with a lot of stimer_t errors. This is directly downloading your branch and trying to compile in Arduino 1.8.13. Any ideas? I'm having a hard time getting marlin working for my printer.

Gunner087, 2021-06-06:
Not sure if this could be your problem, but I had to use version 1.7.0 of the "STM32 Cores" library or it wouldn't compile for the V2 mainboard. I can't remember what the specific error was I got.

Gunner087, 2021-06-07:
Nevermind, just compiled using stm core version 1.9.0 and it worked. @KillerXtreme- is the branch you downloaded the "M200Fixes" branch? If so, that branch does not compile for me either. I would suggest downloading Marlin [here](https://marlinfw.org/meta/download/) and configuring it yourself since the MPSM V2 is now supported in Marlin 2.0.x releases. That's what I did and I didn't have any issues. Just make sure you don't get carried away enabling too many features you don't absolutely need since the STM32 used in the MPSM V2 doesn't have much program storage space and it took me a while to fit the configuration I made for a friend, which included automatic mesh bed leveling with a z-axis probe. You should be able to refer to the branch you downloaded if you are unsure of anything when configuring, not sure how familiar you are with Marlin configuration. If you need any help, let me know. I might have some time available to make a configuration that would be almost complete.

xC0000005, 2021-06-07:
M200 fixes was at a time when we needed some M200 specific code. Now it’s almost as archaic as Malyan’s code (which is also on GitHub).

SharkWipf, 2023-03-06:
Figured I'd contribute back what I've learned trying to get Marlin 2.1 on my brother's M100, since most documentation is a little fragmented and outdated atm. I created a messy-but-hopefully-complete gist of everything I figured out with both this printer and getting Marlin 2.1 working on it.

The tl;dr is however, use upstream Marlin, the work from this repo has been merged into it. A few changes are necessary to get it to build, but nothing too complex.

I'm putting this message here because this was the first resource I found and the resource I kept coming back to most, so I imagine others would as well.

https://gist.github.com/SharkWipf/a9268e78bae29e262421200e64ad4d02

xC0000005, 2023-03-06:
You have a picture of this machine? I never did get the stuff working on the M100, only M200/300 (Mini and Mini pro) machines.

SharkWipf, 2023-03-06:
Whoops, my bad, I meant the M200. Been seeing so many different numbers and names that I've started mixing them up. Updating my gist now.

xC0000005, 2023-03-06:
Also, I see your note about the resulting image being too large for the printer’s flash - regardless of the version, most of them have 128k (it’s possible it’s still too large). So if you want to use the newer STM32, it probably fits. PIO tries to keep you from hitting yourself in the foot, but in this case, it can be handy (and fun).

SharkWipf, 2023-03-06:
>  (it’s possible it’s still too large)

It actually ran out of the full 122880 bytes on Arduino IDE with STM32 2.4, and even the default config on 1.9 (with float printf) uses up 113612 bytes/92%, so it actually just really too big on 2.4.  
I don't know if there's anything in 2.4 that I'd miss over 1.9 anyway, so I'm not too worried about it, as long as 1.9 remains available, though the deprecation warnings all over the library names do look a bit silly.  

I've attached 2 pictures of the printer to the gist as well now, as reply.