xC0000005/Marlin #3: USB Print doesn't trigger printing UI screen

lawrence-jeff, 2019-06-11:
Since both facebook and hackaday are difficult to use for a conversation on a specific topic, opening this issue regarding USB print detection and triggering the build dialog on the UI to prevent 30 minute safety timeout.

My experience is that nothing in the code seems to send {SYS:BUILD} in the USB print scenario so the UI stays on the main page, is this expected?  I struggled to find any specific way to tell if usb print is occuring - so I added some code to check the print timer which is apparently correctly set when a USB print is activated. Sloppy code here https://github.com/lawrence-jeff/Marlin/commit/e5ddd2882a79816e9858d8311143a1b4d9109811

lawrence-jeff, 2019-06-16:
Seems to be addressed by https://github.com/xC0000005/Marlin/commit/9352ae8139d138aee542d8fea3ca97f367ce283e#diff-61a49f76fcd47d1413eb9859a14d138d

xC0000005, 2019-08-14:
Yes - I'm leaving this open until I get the EXTUI PR into Marlin Main.