xC0000005/Marlin #4: Serial output toggles between UI serial and USB serial

lawrence-jeff, 2019-06-11:
I notice serial output seems to output to different ports randomly and I can't seem to identify any pattern. For instance in the LCD code using a SERIAL_ECHOLNPAIR seems to randomly be sent to either the UI (Serial1) or the USB connection (Serial0). Some messages from Marlin itself (like temperature updates) seem to be sent to both.

Opening this to see if you understood why - I dug through the marlin code and it seems these functions should send to Serial0 - but often I get the output on my debug device attached to the UI and it changes based on the function in the LCD code, almost like it just goes to the last serial port written to.

xC0000005, 2019-06-24:
I'm looking at this and not sure - I"m seeing the same thing, and am beginning to think it may be related to using Serial vs USBSerial. I'm creating a PR right now to adapt your changes to the new extui malyan LCD and also fix error output. It now correctly shows when Marlin halts the printer.

xC0000005, 2019-08-14:
I'm working on ExtUI support and once I"m done with that, this is up next on the to-debug list.