xC0000005/Marlin #6: Marlin 2.0.5.3 - S-Curve Acceleration Build failure

theodonkulus, 2020-04-12:
Hey @xC0000005 Just tried to compile in S-Curve acceleration after disabling M503 and found that I'm getting the following assembler compile time errors:

[image]

Do I need to enable some additional FPU support? It doesn't look like the STM32F070 has any hardware acceleration for that.

xC0000005, 2020-04-13:
In platform.txt, we see that we have the line:
compiler.extra_flags=-mcpu={build.mcu} {build.flags.fp} -mthumb "@{build.opt.path}"

Which sets the mode to thumb (I don’t know if this is a starting default or not). You could TRY removing that one option, but I would open an issue on the ST core, this looks wrong.

theodonkulus, 2020-04-13:
I looked at some of the arm optimizatioms as well as I was having trouble fitting stuff in with adding both autolevel + maylan lcd + a few other things.  Seems like thumv reduces instruction sizes

Ill try this tonight and let you know my result. I thought it was odd we were missing instructions but it looks like thumb mode reduces the instruction set?

Oh i also forgot to mention, Im running on the newer V2 systems. Ive actually got 5 of them if you need help smoke test or testing/debugging any of your changes.

 I managed to save 272 bytes on the binary working off your changes and v2.0.5.3 last night :)

xC0000005, 2020-04-13:
The CMSIS docs state that umull (and friends) are not available in thumb mode. To “fix” this we’ll need to figure out what function is being provided in the math file and create a substitute that uses thumb available instructions to do the same. I am thinking the next PR I do will just be a quality of life “#if MB()” fix for the timers. I have a branch with the foundation of picking timers automatically but that’s going to take longer.

xC0000005, 2020-04-14:
What are you using to do analysis? puncover, or nm?

theodonkulus, 2020-04-14:
I just ripped out all your debug with ifdefs and added it part of Configuration.h to be turned on/off when MALYAN_LCD is enabled so far. 

https://github.com/theodonkulus/Marlin/tree/miniV2-maylan-lcd-code-reduce

Its been a while since I used nm so I'll probably take a look at using that again.  Haven't used puncover

I just cherry-picked your M200Fixes2 branch ontop of the v2.0.5.3 tag and then did these changes ontop of that.

theodonkulus, 2020-04-14:
Also on that branch I cant seem to find the thumb option in platformIO.ini. not sure where the other platformIO.txt is...

xC0000005, 2020-04-18:
Can you find me the .s file it's complaining about? It's in the /tmp directory when you hit this. I want to see what the functions are.

theodonkulus, 2020-04-20:
Can't find a .s file that you mentioned but it seems like i might be failing on stepper.cpp as I don't see any object files being output in the modules directory when building this. Still have a .d file I can load here:
[stepper.cpp.d.txt](https://github.com/xC0000005/Marlin/files/4501212/stepper.cpp.d.txt)

Also looking at my platform.ini during build time it seems weird that I don't see an entry for the STM32F07 would that have also something to do with it? I havent used platform IO much. 

[platformio.txt](https://github.com/xC0000005/Marlin/files/4501227/platformio.txt)

Grep output is below. 
theonidus@MahComputer1337 ~/3d_Printing/Marlin (miniV2-maylan-lcd-code-reduce) $ grep -HnIR S_CURVE Marlin/*
Marlin/Configuration.h:826:#define S_CURVE_ACCELERATION
Marlin/src/module/planner.h:131:  #if ENABLED(S_CURVE_ACCELERATION)
Marlin/src/module/planner.h:861:    #if ENABLED(S_CURVE_ACCELERATION)
Marlin/src/module/stepper.h:89:  #if ENABLED(S_CURVE_ACCELERATION)
Marlin/src/module/stepper.h:90:    #define ISR_S_CURVE_CYCLES 40UL
Marlin/src/module/stepper.h:92:    #define ISR_S_CURVE_CYCLES 0UL
Marlin/src/module/stepper.h:119:  #if ENABLED(S_CURVE_ACCELERATION)
Marlin/src/module/stepper.h:120:    #define ISR_S_CURVE_CYCLES 160UL
Marlin/src/module/stepper.h:122:    #define ISR_S_CURVE_CYCLES 0UL
Marlin/src/module/stepper.h:221:#define ISR_EXECUTION_CYCLES(R) (((ISR_BASE_CYCLES + ISR_S_CURVE_CYCLES + (ISR_LOOP_CYCLES) * (R) + ISR_LA_BASE_CYCLES + ISR_LA_LOOP_CYCLES)) / (R))
Marlin/src/module/stepper.h:312:    #if ENABLED(S_CURVE_ACCELERATION)
Marlin/src/module/stepper.h:338:    #if DISABLED(S_CURVE_ACCELERATION)
Marlin/src/module/stepper.h:551:    #if ENABLED(S_CURVE_ACCELERATION)
Marlin/src/module/planner.cpp:255:#if ENABLED(S_CURVE_ACCELERATION)
Marlin/src/module/planner.cpp:784:  #if ENABLED(S_CURVE_ACCELERATION)
Marlin/src/module/planner.cpp:805:    #if ENABLED(S_CURVE_ACCELERATION)
Marlin/src/module/planner.cpp:810:  #if ENABLED(S_CURVE_ACCELERATION)
Marlin/src/module/planner.cpp:815:  #if ENABLED(S_CURVE_ACCELERATION)
Marlin/src/module/planner.cpp:829:  #if ENABLED(S_CURVE_ACCELERATION)
Marlin/src/module/planner.cpp:2251:  #if DISABLED(S_CURVE_ACCELERATION)
Marlin/src/module/stepper.cpp:200:#if ENABLED(S_CURVE_ACCELERATION)
Marlin/src/module/stepper.cpp:231:#if DISABLED(S_CURVE_ACCELERATION)
Marlin/src/module/stepper.cpp:513:#if ENABLED(S_CURVE_ACCELERATION)
Marlin/src/module/stepper.cpp:1316:#endif // S_CURVE_ACCELERATION
Marlin/src/module/stepper.cpp:1675:        #if ENABLED(S_CURVE_ACCELERATION)
Marlin/src/module/stepper.cpp:1702:        #if ENABLED(S_CURVE_ACCELERATION)
Marlin/src/module/stepper.cpp:1960:      #if DISABLED(S_CURVE_ACCELERATION)
Marlin/src/module/stepper.cpp:1965:      #if ENABLED(S_CURVE_ACCELERATION)
Marlin/src/inc/SanityCheck.h:357:  #error "BEZIER_JERK_CONTROL is now S_CURVE_ACCELERATION. Please update your configuration."
Marlin/tags:16895:ISR_S_CURVE_CYCLES	src/module/stepper.h	120;"	d
Marlin/tags:16896:ISR_S_CURVE_CYCLES	src/module/stepper.h	122;"	d
Marlin/tags:16897:ISR_S_CURVE_CYCLES	src/module/stepper.h	90;"	d
Marlin/tags:16898:ISR_S_CURVE_CYCLES	src/module/stepper.h	92;"	d
Marlin/tags:41562:S_CURVE_ACCELERATION	Configuration.h	826;"	d
Marlin/tags:41563:S_CURVE_ACCELERATION	Configuration_bak.h	805;"	d